pi1541/Pi1541 #226: Any plan for a Pi0 2 port?

andreacampanella, 2021-12-06:
Maybe I missed in my research but I was wondering if there is any plan to port it to the Pi zero 2? 
I've been out of the loop for a while so I don't really know if there is going to be any performance improvement over the normal one.

gestorum, 2021-12-31:
Works for me on my RPi zero 2 using a new build from v1.24 source code but I first need to disable my Epyx Fastload cartridge to be able to load files properly.

EddyBeaupre, 2022-03-10:
Made it wirk on my Pi Zero 2w with Epyx FastLoad. Here's what i had to do:

1. Install Raspberry OS on a spare SD and boot your Pi Zero 2W with it. 
2. Update everything (apt update && apt dist-upgrade -y)
3. Install the toolchain to compile a Pi1541 Kernel.
4. Compile a Kernel on the Pi Zero 2W itself (tested with v1.24 and master).
5. Use "bootcode.bin", "fixup.dat" and "start.elf" from the /boot partition of  the SD.
6. Add "arm_freq=1200" to the Pi1541 "config.txt", my Pi is stable without any over-voltage, but you can also add " over_voltage=6" to be on the safe side.
7. Comment out "GraphIEC = 1" in options.txt if set.
8. I also had to disable the buzzer, it sometime got stuck with a tone, but the Pi1541 was working well with it.

Your milleage may definitly vary....

gestorum, 2022-03-10:
The config.txt tweak did the job. Thanks for your help.

SteveGuidi, 2024-06-16:
> Made it wirk on my Pi Zero 2w with Epyx FastLoad. Here's what i had to do:
>
> 1. Install Raspberry OS on a spare SD and boot your Pi Zero 2W with it.
> 2. Update everything (apt update && apt [...]

I'm about to start on this project in the upcoming weeks and wanted to ask if the ancillary Pi Zero 2w steps are still needed.  I found @EddyBeaupre's [github page](https://github.com/Beaupre-Circuit-Design/c64-Pi1541-Zero2W-Firmware) that nicely packages the above steps, but the main [Pi1541](https://cbm-pi1541.firebaseapp.com) page now provides Pi Zero 2(but not W)  firmware.