pi1541/Pi1541 #229: Cannot get ROMchar / chargen to work on Pi0

DeeKay64, 2022-02-07:
I have a Pi0 with SH1106 I2C display, and i just cannot seem to get the ROMchar (which would give me twice as many lines since it's just 8px high!) to work. I told it the filename of the ROMchar via ChargenFont and set i2cLcdUseCBMChar = 1 in the options, but nothing seems to happen...

ROMchar works fine on my standard Pi1541 with a SSD1306 display running on a 3A...

I remember someone on some forum saying they also couldn't get this to work on Pi0. What is the issue exactly?

DeeKay64, 2022-02-17:
Steve told me this on FB:
"Hey Daniel, it appears you are correct. For some unknown reason the code for loading the char set is excluded for the Pi0. It will be fixed in the next version." (that's 1.25!)

tactaf, 2022-04-03:
> Steve told me this on FB: "Hey Daniel, it appears you are correct. For some unknown reason the code for loading the char set is excluded for the Pi0. It will be fixed in the next version." (that's 1.25!)

Having the same issue as you with a PI0, except I am using a SH1106 LCD. Same Hat/LCD with 3A, no issues.  Went hunting for an answer,  thought I had better check github first.  This was the answer I was looking for. Thanks.

john-clark, 2024-01-16:
Can verify that the chargen is working on pi3 and then switch 1.24 kernel and put the hat on a pi1a+ and it does not work.

It has been several years now since the pi0/1 is not working. Any chance for that 1.25 code?