pi1541/Pi1541 #231: Can´t load varios Demos on V1.24 (V1.23 works fine)

Snocksman, 2022-02-21:
Hi !

I tested to load several Demos with Version 1.24 and found many of them that stop loading / crash with V1.24. The same Demos work fine with Version 1.23.

Here are two of them:
Comaland ( link: https://csdb.dk/release/?id=139278 )
Edge of Disgrace ( link: https://csdb.dk/release/?id=72550 )

elerion64, 2022-03-27:
I can confirm. 
Also, Nostalgia release of Ghosts and Goblins shows garbled first screen with pi0 and V1.24.

liaminventions, 2022-07-27:
confirmed

LazeMSS, 2022-12-27:
I can also confirm this - I was wondering if it could be related to the rpi core firmware?

h3r3, 2023-02-22:
I'm so glad I found this!
Basically all the games and demos I have didn't work with 1.24 and I thought something was wrong with my C64.
Changed the firmware to 1.23 and everything works!
I wish there was at least a note on the website about the problems with 1.24.