pi1541/Pi1541 #238: OLED screen doesn't display

ajkatz83, 2022-07-25:
I have been using a standard I2C OLED display. In the the option.txt I turned on the scan feature to see if it  would pick up and devices and it has on I2C master 1, Address 60, yet nothing displays. I have 5V going to it also, and nothing still. I know it cannot be this difficult to activate an OLED. This always seems to be my luck that everything else works, but the one finer details just never do. I f there is any advice on this please let me know.

faxm0dem, 2023-02-18:
Hi,

I have a similar setup, and OLED screen is switching on at power up, with the 1541 Logo and version.
It stays for a few seconds then switches off again.

What is it supposed to display after that ?

john-clark, 2023-12-17:
This worked for me in `options.txt`

```ini
splitIECLines = 1
LCDName = ssd1306_128x64
LcdLogoName = 1541classic
i2cBusMaster = 1
```