pi1541/Pi1541 #243: Setting hdmi screen resolution not working.`

lukolszewski, 2022-11-25:
I have a waveshare 7in hdmi lcd screen for my rpi 3. It requires 1024x600 resolution. However when I set the 
ScreenWidth = 1024
ScreenHeight = 600
It still runs at a resolution my 1024x600 screen doesn't support. When I plug it into a full size monitor with a hdmi cable it definitely looks like it is still running at its default resolution.

Is it a bug? Is the resolution setting just for composite? If yes, is there someone in the source the hdmi output resolution is set?

vejinnovem, 2022-11-25:
https://www.raspberrypi.com/documentation/computers/config_txt.html
look for "hdmi_mode" section

Deadweasel, 2023-07-19:
I have a UCTronics 7" screen that has interesting limitations as well.  The vendor supplies a .sh file that you're supposed to run in order to make the screen work, but in reality all the script is doing is adding the following to the boot\config.txt of Raspbian:
hdmi_force_hotplug=1
max_usb_current=1 
hdmi_group=2
hdmi_driver=2
hdmi_mode=16

Resolution is technically 1024x600, but depending on your tolerance for scaling shenanigans, you can also get away with 1280x720 and yes, even 1920x1080, if you have superhuman vision.  :D