pi1541/Pi1541 #248: Incorrect status returned after error

kirbyfrugia, 2023-08-27:
When I do the following:
1. Load a file
2. Save the file (which already exists). Do not add "@:" for this example.
3. LED starts flashing. Error channel correctly shows an error code 63. However, the c64 just shows READY instead of displaying the error.

Here are some videos to see the issue in action:
- [The error occurring](https://share.icloud.com/photos/084D0TXgzPP5hQyILIsawL-Wg)
- [Reading the error channel](https://share.icloud.com/photos/0f39siur_78GN6BjeTjl5Mk4A)
- [Printing ST](https://share.icloud.com/photos/0a5Bl6EydJR9yUq97fFobpa4w)

radius75, 2023-08-27:
Have you checked the same on a real 1541 Disk Drive?

kirbyfrugia, 2023-08-27:
Oh wow. I just tried with my real 1541-II. Same issue! I guess it isn't the pi1541. Any thoughts on what might cause this? I realize it's off topic now and this issue can be closed.

radius75, 2023-08-27:
This is normal C64 and 1541 behavior.
You can close the thread yourself.
The author of the project probably took a long vacation or abandoned the project.

kirbyfrugia, 2023-08-27:
Strange. I swear I have seen diff behavior in the past but maybe I am misremembering.

closed 2023-08-27 by kirbyfrugia