pi1541/Pi1541 #257: Won't load files saved with the Evesham Micros Freeze Machine

ajt-en-france, 2024-12-01:
I've a number of 5.25 disks which have various programs on them which were frozen and then saved with Evesham Micros Freeze Machine. They still work perfectly well on my Oceanic OC-118 floppy drive.

I can copy the disks with DRA Copy to a d64 file with a Pi1541 on a Raspberry PI 3. The files will load perfectly under VICE on my PC but never from the Pi1541 into my real C64. VICE and the Pi1541 are using the same ROM images (1541-II).

This isn't a critical problem but I'd like to be able to use straight copies of my files if that were possible. The Freeze Machine Cartridge isn't installed in either my real C4 or the VICE emulator and the disks work fine via the real drive and under VICE.

Can provide d64 files if useful.

UffeJakobsen, 2024-12-01:
What version of the pi1541 firmware are you running ?

ajt-en-france, 2024-12-01:
It's the 1.24 version directly downloaded from the https://cbm-pi1541.firebaseapp.com/

I know that there are beta versions available and some people report that the 1.23 can sometimes be more stable. I've not tried an older version yet.

I think the Evesham Freeze Machine wasn't common outside of the UK, and so it's probably not been tested with a lot.

UffeJakobsen, 2024-12-01:
I would try version 1.23 - just to be sure

ajt-en-france, 2024-12-02:
I can confirm that with the 1.23 kernel it does indeed load the files as expected just like the real drive does or VICE. It seems like another minor problem with the 1.24 version.

Should anyone wish to test I can make a D64 available to test with.

closed 2024-12-03 by ajt-en-france

UffeJakobsen, 2024-12-03:
> 
> Should anyone wish to test I can make a D64 available to test with.

@ajt-en-france yes please - could you make the D64 available ?

ajt-en-france, 2024-12-03:
I can make the file(s) available. It's all "back-ups" of my own games originally on tape, and is copyright material. I know the content is easily found on-line but I wouldn't want anyone to get into trouble here. How would you like it?

UffeJakobsen, 2024-12-03:
Hmm good question - as I believe the none of us wants to leave an email in the open here for the world to use for spamming. :-)
What I was looking for was a D64 that worked on 1.23 and didn't work in 1.24... and I believe that I've qualified a few other D64's - so I guess that we should just close this case (but thanks for your helpfulness)  :-)

ajt-en-france, 2024-12-03:
If you do want a .d64 file, I'm willing to place one on my own server and make it clear how to get to it from here without posting anything here that may be considered a copyright violation. I can also have a go at creating a new blank image with the cartridge and and using something that I own the copyright of personally so would be no violation - that I could simply attach here.

I've seen several people comment that the 1.24 version has a few issues not found in the 1.23 version, so I'm happy to use the 1.23 version for now - I've been able to transfer most of my disk to d64 files - not that it would be a great loss to the world if I hadn't.

**To everyone who helped make this happen I'm most impressed...

UffeJakobsen, 2024-12-03:
Thanks for your offer - In the meantime I've found another D64 image suitable for compatibility testing - see #256 - I think that is good for now - thanks :-)