pi1541/Pi1541 #258: [FEATURE REQUEST] : HDMI output format options

Deadweasel, 2024-12-21:
Got the HDMI output working fine and fullscreen on my little cheapie 7" IPS screen, but was hoping to find a way to enlarge the font being used somehow.  As it shows on the screen at present, I can mostly make out what's being displayed, but I have to get up close and squint a tad to be able to decipher the content.

Would help if there was a way to enlarge the text?

penfold42, 2024-12-21:
Have you tried fiddling with HDMI modes as mentioned in #243 ?

It wont change the font size but you could drop the resolution of the display.

Deadweasel, 2024-12-23:
> Have you tried fiddling with HDMI modes as mentioned in #243 ?
> 
> It wont change the font size but you could drop the resolution of the display.

Well I am able to get the screen actually working with the settings I called out in that reply, but it's pretty fiddly and does not really display much different than perhaps the aspect ratio changes when different resolutions are attempted, but they are all simply scaled up/down to the monitor's own expected resolution, regardless.
In this case, it's not as simple as using the OS display controls to select a lower resolution, because there are no fancy drivers to perform some of the pre-output trickery that's available with them, like font scaling or screen magnification etc, so higher or lower resolution settings don't matter; the screen is just going to either stay blank or scale the output to its preferred resolution anyway.

Incidentally, I have found this same behavior applies to any display I've connected, be it a Dell 24" monitor or Vizio 34" TV.  They will all show the same relative text size; the only thing that makes it larger or smaller is the size of the screen being used to display the output.

I'm not really much of a C coder, but I'm guessing the text size is being determined by scaling it based on the current output resolution, which is common for other languages.  If that's the case, I'd love to have even a keyboard shortcut to adjust that scaling up and down.